Biktorgj/quectel_eg25_recovery #1: did the upgrade work? dmesg still reports outdated firmware

lazmol, 2020-12-24:
Hello,
I have recently upgraded the motherboard of my ubports edition pinephone and tried to upgrade the firmware of the modem.
I have unlcoked it following [this page](https://xnux.eu/devices/feature/modem-pp.html#toc-unlock-adb-access) by compiling and running the qadbkey-unlock c code with 23534831, which gave me the output

```
AT+QADBKEY="YzARSWZrjs1xdASl"
AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,1,0

To disable ADB, run: (beware that modem will not be able to enter sleep with ADB enabled!!)
AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,0,0
```

Ran the first two commands after running `sudo picocom -b 115200 /dev/ttyUSB2`
I have restarted the modem in edl with `sudo adb reboot edl` and confirmed with lsusb that it is in QDL mode, then ran the upgrade with: `sudo ./qfirehose_arm64 -f ./ > upgrade.log`
The end of the output was (I could not attach a file, but I can send if you want):

```
[032.345]: Upgrade progress:   100
[032.347]: send finished
[032.348]: <log value="Finished sector address 13952"/>
[032.348]: <response value="ACK" rawmode="false" />
[032.348]: <power value="reset" />
[032.349]: <log value="Inside handlePower() - Requested POWER_RESET"/>
[032.349]: <response value="ACK" />
[033.351]: inf[0] ep_in -1/1024, errno = 108 (Cannot send after transport endpoint shutdown)
[033.351]: qusb_noblock_read read=-1, errno: 108 (Cannot send after transport endpoint shutdown)
[033.351]: qusb_noblock_read cur=0, min_size=1
[033.351]: firehose_protocol.c fh_recv_cmd 309 fail
[033.351]: THE TOTAL DOWNLOAD TIME IS 29.305 s
[033.351]: Upgrade module successfully.
```

I've just cloned your repo, so I was on the default branch (the one with 01.003 in the end).
Waited for the modem to come up, rebooted the phone, ran `journalctl -xb --no-pager | grep modem > modem.log`, still shows that it is outdated:

```
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: ===================================================
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Project Name: EG25G
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Project Rev : EG25GGBR07A08M2G_01.003
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Branch  Name: 9x07_R07_NEW
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Custom Name : STD
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Build   Date: 2020-11-09,18:01
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Version:        SBLR07EG25G01_1:37500: May 30 2
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Version:        TZR07A03: Dec  2 2019
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Version:        RPMR07A03_1:37500: Dec  2 2019
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: ===================================================
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: Your modem has an outdated firmware. Latest know version is EG25GGBR07A08M2G_01.002.07. Consider updating.
Dec 24 13:18:22 lacophone kernel: modem-power serial1-0: ADB KEY is '23534831' (you can use it to unlock ADB access to the modem, see https://xnux.eu/devices/feature/modem-pp.html)
```
Again, can send all kernel messages with 'modem'.

Thanks for any help in advance!

Biktorgj, 2020-12-25:
Your firmware has been updated, it's just the modem kernel driver looking for another version. There have been some reports of some issues during sleep with the latest firmware, so you might actually want to downgrade to EG25GGBR07A08M2G_01.002.07, which looks more stable.

lazmol, 2020-12-26:
Thanks @Biktorgj ! How to downgrade to version EG25GGBR07A08M2G_01.002.07? I do not see such a branch. Or is it the EG25GGBR07A08M2G_01.002.01.002?

Biktorgj, 2020-12-27:
I thought I had pushed it, sorry
https://github.com/Biktorgj/quectel_eg25_recovery/tree/EG25GGBR07A08M2G_01.002.07

Let me know if it works for you

lazmol, 2020-12-27:
> I thought I had pushed it, sorry
> https://github.com/Biktorgj/quectel_eg25_recovery/tree/EG25GGBR07A08M2G_01.002.07
> 
> Let me know if it works for you

Cheers! Worked like a charm, kernel message is gone too.
I close this issue (will use as a reference for a future upgrade). Thx again!

closed 2020-12-27 by lazmol