Biktorgj/quectel_eg25_recovery #2: Can not reboot into EDL mode, no permissions for device

FalcoGer, 2021-01-20:
I get the following issue when trying to follow the steps in the readme:

```text
./qadbkey-unlock ****
AT+QADBKEY="****"
AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,1,0

To disable ADB, run: (beware that modem will not be able to enter sleep with ADB enabled!!)
AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,0,0
manjaro@pinephone:/home/manjaro/repositories$ sudo mmcli -m any --command='AT+QADBKEY="****"'
response: ''
manjaro@pinephone:/home/manjaro/repositories$ sudo mmcli -m any --command='AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,1,0'
response: ''
manjaro@pinephone:/home/manjaro/repositories$ adb reboot edl
error: insufficient permissions for device
See [http://developer.android.com/tools/device.html] for more information
manjaro@pinephone:/home/manjaro/repositories$ sudo adb reboot edl
error: insufficient permissions for device
See [http://developer.android.com/tools/device.html] for more information
manjaro@pinephone:/home/manjaro/repositories$ adb devices
List of devices attached
(no serial number)	no permissions; see [http://developer.android.com/tools/device.html]
```

Biktorgj, 2021-01-20:
You need to run adb commands as root, as the manjaro user doesn't have enough permissions. Kill adb (adb kill-server or killall -9 adb), and try the commands again, but this time as root (sudo adb reboot edl)

FalcoGer, 2021-01-20:
I ran the update without edl mode and it seemed to have updated the modem. (GPS still doesn't work though... oh well.)

closed 2021-01-20 by FalcoGer