Biktorgj/quectel_eg25_recovery #5: can't upgrade to EG25GGBR07A08M2G_01.003.01.003

BangL, 2021-06-26:
board: v1.2b
distro: archlinux-pinephone-phosh-20210613.img

entered EDL mode by adb:
> [root@danctnix src]# ./adb reboot edl          
[root@danctnix src]# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 004: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

but the upgrade process fails:
> [root@danctnix quectel_eg25_recovery]# ./qfirehose_arm64 -f ./
[000.000]: QFirehose Version: Quectel_LTE&5G_QFirehose_Linux&Android_V1.2.3
[000.000]: Builded: Jul 31 2020 11:11:55
[000.001]: Find md5 check file <./md5.txt>
[000.002]: md5 checking: ./contents.xml pass
[000.026]: md5 checking: ./update/appsboot.mbn pass
[000.032]: md5 checking: ./update/ENPRG9x07.mbn pass
[000.033]: md5 checking: ./update/firehose/partition_complete_p2K_b128K.mbn pass
[000.033]: md5 checking: ./update/firehose/patch_p2K_b128K.xml pass
[000.043]: md5 checking: ./update/firehose/prog_nand_firehose_9x07.mbn pass
[000.044]: md5 checking: ./update/firehose/rawprogram_nand_p2K_b128K_update.xml pass
[000.253]: md5 checking: ./update/mdm9607-boot.img pass
[000.554]: md5 checking: ./update/mdm9607-recovery.ubi pass
[002.233]: md5 checking: ./update/mdm9607-sysfs.ubi pass
[003.452]: md5 checking: ./update/NON-HLOS.ubi pass
[003.455]: md5 checking: ./update/NPRG9x07.mbn pass
[003.455]: md5 checking: ./update/partition.mbn pass
[003.455]: md5 checking: ./update/partition_nand.xml pass
[003.460]: md5 checking: ./update/rpm.mbn pass
[003.467]: md5 checking: ./update/sbl1.mbn pass
[003.483]: md5 checking: ./update/tz.mbn pass
[003.546]: md5 checking: ./update/usrdata.ubi pass
[003.546]: Totals checking 18 files md5 value, 0 file fail!
[003.547]: [1] /sys/bus/usb/devices/2-1 5c6/9008/0
[003.548]: P: /dev/bus/usb/002/004 idVendor=05c6 idProduct=9008
[003.548]: C: /dev/bus/usb/002/004 bNumInterfaces: 1
[003.548]: I: If#= 0 Alt= 0 #EPs= 2 Cls=ff Sub=ff Prot=ff
[003.548]: E: Ad=81 Atr=02 MxPS= 512 Ivl=0ms
[003.548]: E: Ad=01 Atr=02 MxPS= 512 Ivl=0ms
[003.549]: usbfs_is_kernel_driver_alive find interface 0 has match the driver qcserial
[003.551]: usbfs_detach_kernel_driver detach kernel driver success
[003.554]: dir=./update/firehose
[003.554]: d_name=prog_nand_firehose_9x07.mbn
[003.554]: prog_nand_firehose_filename = prog_nand_firehose_9x07.mbn
[003.554]: STATE <-- SAHARA_WAIT_HELLO
[003.605]: inf[0] ep_in -1/4096, errno = 108 (Cannot send after transport endpoint shutdown)
[003.605]: qusb_noblock_read read=-1, errno: 108 (Cannot send after transport endpoint shutdown)
[003.605]: qusb_noblock_read cur=0, min_size=1
[003.605]: inf[0] USBDEVFS_SUBMITURB -1/1, errno = 19 (No such device)
[003.605]: qusb_noblock_write write=-1, errno: 19 (No such device)
[003.605]: qusb_noblock_write cur=0, min_size=1
[003.605]: inf[0] ep_in -1/4096, errno = 19 (No such device)
[003.605]: qusb_noblock_read read=-1, errno: 19 (No such device)
[003.605]: qusb_noblock_read cur=0, min_size=1
[003.605]: Sahara protocol error
[003.606]: Upgrade module failed.

what am i doing wrong? :(

Biktorgj, 2021-06-26:
It looks like something is messing with the modem while in EDL forcing it either to shutdown or reboot. When it fails, what is the output of `lsusb`?

By the way, version 003.01 is buggy as hell, so, except for giving it a shot, I wouldn't recommend you to stay on that (that's why the default branch in the repo is 01.002.01.002)

BangL, 2021-06-26:
switched to 01.002.01.002 now and tried it again. same problem. edl mode stais.

> [root@danctnix ~]# screen /dev/ttyUSB2 115200
[screen is terminating]
[root@danctnix ~]# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[root@danctnix ~]# cd /home/alarm/arm_adb/src/
[root@danctnix src]# ./sdb reboot edl
-bash: ./sdb: Datei oder Verzeichnis nicht gefunden
[root@danctnix src]# ./adb reboot edl 
* daemon not running; starting now at tcp:5038
* daemon started successfully
[root@danctnix src]# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[root@danctnix src]# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[root@danctnix src]# cd ..
[root@danctnix arm_adb]# cd ..
[root@danctnix alarm]# cd ..
[root@danctnix home]# cd alarm/quectel_eg25_recovery/
[root@danctnix quectel_eg25_recovery]# ls -lah
insgesamt 3,1M
drwxr-xr-x  4 alarm alarm 4,0K 26. Jun 16:39 .
drwx------ 19 alarm alarm 4,0K 26. Jun 16:39 ..
-rw-r--r--  1 alarm alarm 202K 26. Jun 03:13 board.jpg
-rwxr-xr-x  1 alarm alarm 4,8K 26. Jun 03:13 contents.xml
drwxr-xr-x  8 alarm alarm 4,0K 26. Jun 16:39 .git
-rw-r--r--  1 alarm alarm 3,1K 26. Jun 03:13 install.sh
-rwxr-xr-x  1 alarm alarm 1,3K 26. Jun 16:39 md5.txt
-rwxr-xr-x  1 alarm alarm 1,1M 26. Jun 03:13 qfirehose
-rwxr-xr-x  1 alarm alarm 121K 26. Jun 03:13 qfirehose_arm64
-rwxr-xr-x  1 alarm alarm 121K 26. Jun 03:13 qfirehose_arm64_ld
-rwxr-xr-x  1 alarm alarm  66K 26. Jun 03:13 qfirehose_armhf
-rwxr-xr-x  1 alarm alarm 721K 26. Jun 16:39 Quectel_EG25-G-GB_Firmware_Release_Notes_V0708_01.002.01.002.pdf
-rwxr-xr-x  1 alarm alarm 738K 26. Jun 16:39 Quectel_EG25-G-GB_软件版本变更说明_V0708.01.002.01.002.pdf
-rw-r--r--  1 alarm alarm 1,3K 26. Jun 16:39 README.md
drwxr-xr-x  3 alarm alarm 4,0K 26. Jun 16:39 update
-rw-r--r--  1 alarm alarm  153 26. Jun 03:13 update.service
[root@danctnix quectel_eg25_recovery]# ./qfirehose -f ./
[000.000]: QFirehose Version: Quectel_LTE&5G_QFirehose_Linux&Android_V1.2.3
[000.000]: Builded: Oct 21 2020 19:13:31
[000.001]: Find md5 check file <./md5.txt>
[000.003]: md5 checking: ./contents.xml pass
[000.032]: md5 checking: ./update/appsboot.mbn pass
[000.046]: md5 checking: ./update/ENPRG9x07.mbn pass
[000.049]: md5 checking: ./update/firehose/partition_complete_p2K_b128K.mbn pass
[000.050]: md5 checking: ./update/firehose/patch_p2K_b128K.xml pass
[000.059]: md5 checking: ./update/firehose/prog_nand_firehose_9x07.mbn pass
[000.060]: md5 checking: ./update/firehose/rawprogram_nand_p2K_b128K_update.xml pass
[000.290]: md5 checking: ./update/mdm9607-boot.img pass
[000.697]: md5 checking: ./update/mdm9607-recovery.ubi pass
[002.736]: md5 checking: ./update/mdm9607-sysfs.ubi pass
[004.057]: md5 checking: ./update/NON-HLOS.ubi pass
[004.069]: md5 checking: ./update/NPRG9x07.mbn pass
[004.071]: md5 checking: ./update/partition.mbn pass
[004.072]: md5 checking: ./update/partition_nand.xml pass
[004.084]: md5 checking: ./update/rpm.mbn pass
[004.099]: md5 checking: ./update/sbl1.mbn pass
[004.125]: md5 checking: ./update/tz.mbn pass
[004.210]: md5 checking: ./update/usrdata.ubi pass
[004.210]: Totals checking 18 files md5 value, 0 file fail!
[004.211]: [1] /sys/bus/usb/devices/2-1 5c6/9008/0
[004.213]: P: /dev/bus/usb/002/003 idVendor=05c6 idProduct=9008
[004.213]: C: /dev/bus/usb/002/003 bNumInterfaces: 1
[004.213]: I: If#= 0 Alt= 0 #EPs= 2 Cls=ff Sub=ff Prot=ff
[004.213]: E: Ad=81 Atr=02 MxPS= 512 Ivl=0ms
[004.213]: E: Ad=01 Atr=02 MxPS= 512 Ivl=0ms
[004.213]: usbfs_is_kernel_driver_alive find interface 0 has match the driver qcserial
[004.215]: usbfs_detach_kernel_driver detach kernel driver success
[004.223]: dir=./update/firehose
[004.224]: d_name=prog_nand_firehose_9x07.mbn
[004.224]: prog_nand_firehose_filename = prog_nand_firehose_9x07.mbn
[004.224]: STATE <-- SAHARA_WAIT_HELLO
[004.262]: inf[0] ep_in -1/4096, errno = 108 (Cannot send after transport endpoint shutdown)
[004.263]: qusb_noblock_read read=-1, errno: 108 (Cannot send after transport endpoint shutdown)
[004.263]: qusb_noblock_read cur=0, min_size=1
[004.272]: inf[0] USBDEVFS_SUBMITURB -1/1, errno = 19 (No such device)
[004.273]: qusb_noblock_write write=-1, errno: 19 (No such device)
[004.273]: qusb_noblock_write cur=0, min_size=1
[004.273]: inf[0] ep_in -1/4096, errno = 19 (No such device)
[004.273]: qusb_noblock_read read=-1, errno: 19 (No such device)
[004.273]: qusb_noblock_read cur=0, min_size=1
[004.273]: Sahara protocol error
[004.273]: Upgrade module failed.
[root@danctnix quectel_eg25_recovery]# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

BangL, 2021-06-26:
additional note:
yesterday, before i posted the `19 (no such device)` error, i had a few tries before with different logs.
could these have been destroying my edl mode?

here are the older logs:
```
[alarm@danctnix quectel_eg25_recovery]$ sudo ./qfirehose -f ./
[sudo] Passwort für alarm: 
[000.000]: QFirehose Version: Quectel_LTE&5G_QFirehose_Linux&Android_V1.2.3
[000.001]: Builded: Oct 21 2020 19:13:31
[000.001]: Find md5 check file <./md5.txt>
[000.001]: md5 checking: ./contents.xml pass
[000.019]: md5 checking: ./update/appsboot.mbn pass
[000.023]: md5 checking: ./update/ENPRG9x07.mbn pass
[000.023]: md5 checking: ./update/firehose/partition_complete_p2K_b128K.mbn pass
[000.023]: md5 checking: ./update/firehose/patch_p2K_b128K.xml pass
[000.027]: md5 checking: ./update/firehose/prog_nand_firehose_9x07.mbn pass
[000.028]: md5 checking: ./update/firehose/rawprogram_nand_p2K_b128K_update.xml pass
[000.213]: md5 checking: ./update/mdm9607-boot.img pass
[000.518]: md5 checking: ./update/mdm9607-recovery.ubi pass
[002.211]: md5 checking: ./update/mdm9607-sysfs.ubi pass
[003.440]: md5 checking: ./update/NON-HLOS.ubi pass
[003.444]: md5 checking: ./update/NPRG9x07.mbn pass
[003.444]: md5 checking: ./update/partition.mbn pass
[003.444]: md5 checking: ./update/partition_nand.xml pass
[003.450]: md5 checking: ./update/rpm.mbn pass
[003.457]: md5 checking: ./update/sbl1.mbn pass
[003.473]: md5 checking: ./update/tz.mbn pass
[003.538]: md5 checking: ./update/usrdata.ubi pass
[003.538]: Totals checking 18 files md5 value, 0 file fail!
[003.539]: [1] /sys/bus/usb/devices/2-1 5c6/9008/0
[003.540]: P: /dev/bus/usb/002/002 idVendor=05c6 idProduct=9008
[003.540]: C: /dev/bus/usb/002/002 bNumInterfaces: 1
[003.540]: I: If#= 0 Alt= 0 #EPs= 2 Cls=ff Sub=ff Prot=ff
[003.540]: E: Ad=81 Atr=02 MxPS= 512 Ivl=0ms
[003.540]: E: Ad=01 Atr=02 MxPS= 512 Ivl=0ms
[003.540]: usbfs_is_kernel_driver_alive find interface 0 has match the driver qcserial
[003.542]: usbfs_detach_kernel_driver detach kernel driver success
[003.544]: dir=./update/firehose
[003.545]: d_name=prog_nand_firehose_9x07.mbn
[003.545]: prog_nand_firehose_filename = prog_nand_firehose_9x07.mbn
[003.545]: STATE <-- SAHARA_WAIT_HELLO
[008.741]: inf[0] ep_in -1/4096, errno = 110 (Connection timed out), timeout=5000
[008.741]: qusb_noblock_read read=0, errno: 110 (Connection timed out)
[008.741]: qusb_noblock_read cur=0, min_size=1
[008.741]: RECEIVED <-- SAHARA_END_IMAGE_TX_ID
[008.742]: Received a different command: 4 while waiting for hello packet
[008.742]: SENDING --> SAHARA_RESET
[008.742]: Sahara protocol error
[008.742]: Upgrade module failed.
[alarm@danctnix quectel_eg25_recovery]$ lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[alarm@danctnix quectel_eg25_recovery]$ sudo -i
[root@danctnix ~]# cd /home/alarm/quectel_eg25_recovery/
[root@danctnix quectel_eg25_recovery]# ./qfirehose -f ./
[000.000]: QFirehose Version: Quectel_LTE&5G_QFirehose_Linux&Android_V1.2.3
[000.000]: Builded: Oct 21 2020 19:13:31
[000.000]: Find md5 check file <./md5.txt>
[000.001]: md5 checking: ./contents.xml pass
[000.026]: md5 checking: ./update/appsboot.mbn pass
[000.031]: md5 checking: ./update/ENPRG9x07.mbn pass
[000.036]: md5 checking: ./update/firehose/partition_complete_p2K_b128K.mbn pass
[000.036]: md5 checking: ./update/firehose/patch_p2K_b128K.xml pass
[000.042]: md5 checking: ./update/firehose/prog_nand_firehose_9x07.mbn pass
[000.042]: md5 checking: ./update/firehose/rawprogram_nand_p2K_b128K_update.xml pass
[... 51 lines cut ...]
[003.469]: md5 checking: ./update/tz.mbn pass
[003.532]: md5 checking: ./update/usrdata.ubi pass
[003.532]: Totals checking 18 files md5 value, 0 file fail!
[003.533]: [1] /sys/bus/usb/devices/2-1 5c6/9008/0
[003.534]: P: /dev/bus/usb/002/002 idVendor=05c6 idProduct=9008
[003.534]: C: /dev/bus/usb/002/002 bNumInterfaces: 1
[003.534]: I: If#= 0 Alt= 0 #EPs= 2 Cls=ff Sub=ff Prot=ff
[003.534]: E: Ad=81 Atr=02 MxPS= 512 Ivl=0ms
[003.534]: E: Ad=01 Atr=02 MxPS= 512 Ivl=0ms
[003.534]: usbfs_is_kernel_driver_alive ioctl USBDEVFS_GETDRIVER failed, kernel driver may be inactive
[003.536]: dir=./update/firehose
[003.536]: d_name=prog_nand_firehose_9x07.mbn
[003.536]: prog_nand_firehose_filename = prog_nand_firehose_9x07.mbn
[003.536]: STATE <-- SAHARA_WAIT_HELLO
[003.541]: inf[0] ep_in -1/4096, errno = 71 (Protocol error)
[003.541]: qusb_noblock_read read=-1, errno: 71 (Protocol error)
[003.541]: qusb_noblock_read cur=0, min_size=1
[003.545]: qusb_noblock_write write=-1, errno: 71 (Protocol error)
[003.545]: qusb_noblock_write cur=0, min_size=1
[003.549]: inf[0] ep_in -1/4096, errno = 71 (Protocol error)
[003.549]: qusb_noblock_read read=-1, errno: 71 (Protocol error)
[003.549]: qusb_noblock_read cur=0, min_size=1
[003.549]: Sahara protocol error
[003.549]: Upgrade module failed.
```

BangL, 2021-06-26:
oh and 2 more things, possibly worth noting:
1. my current ArchARM kernel does not seem to expose the `modem-power` thing. could that be a problem?
2. when i reboot normally, my modem still works fine for now (also just received a call). i just have no clue what version i'm on.

BangL, 2021-07-04:
solved it by retrying it using latest mobian nightly.
looks like it was a problem with latest arch-arm

closed 2021-07-04 by BangL