Biktorgj/quectel_eg25_recovery #6: Some notes about firmware versions

Zapeth, 2021-07-05:
I was curious about all the different firmware versions that were available online, so I compiled a list of my findings:

| fw-version | modem-timestamp | modem-sha1-hash (NON-HLOS.ubi) | source |
| ------- |:-----------:|:----------:|:-----------:|
| EG25GGBR07A07M2G_01.001.02 | 2019-04-15 10:15:59 | 4DA6394867589A35D6EE2393AF08F28E1EE8F6F6 | [ftp link](https://ftp1.digi.com/support/firmware/dal/carrier_firmware/EG25-G%20(1002-CMG4)/Archived%20Firmware/EG25GGBR07A07M2G.tar.gz) |
| EG25GGBR07A07M2G_01.002.07 | 2019-08-21 19:47:17 | A39571B80023CD77249E8D4756B43D227D22A855 | [master branch](https://github.com/Biktorgj/quectel_eg25_recovery/tree/master) |
| EG25GGBR07A08M2G_01.001.10 | 2019-12-03 18:17:59 | | stock firmware (Pinephone pmOS CE, dont know about current Pinephone Beta Edition) |
| EG25GGBR07A08M2G_01.002.07 | 2020-06-13 09:32:06 | 6913F5F2B3A6F3820935DC4E21E713E7CE396747 | [EG25GGBR07A08M2G_01.002.07 branch](https://github.com/Biktorgj/quectel_eg25_recovery/tree/EG25GGBR07A08M2G_01.002.07)*
| EG25GGBR07A08M2G_01.003.04 | 2020-11-09 18:15:59 | 1E03160F229451FC09DAC079F7027B9BDAFC9A7A | [EG25GGBR07A08M2G_01.003.01.003 branch](https://github.com/Biktorgj/quectel_eg25_recovery/tree/EG25GGBR07A08M2G_01.003.01.003) |

\* ... binary identical to [EG25GGBR07A08M2G_01.002.01.002 branch](https://github.com/Biktorgj/quectel_eg25_recovery/tree/EG25GGBR07A08M2G_01.002.01.002) (except this one also includes the pdfs about the release changes)

I couldn't find any flashable files for the stock version online, so I tried dumping them from my phone. However I was only able to get the ubifs volumes for `rootfs` and `modem`, I'm not sure how `boot` and `recovery` partitions can be dumped (if at all). And even those dumped ubifs volumes seem to include unallocated data, so I'm not sure if those are safe for flashing.

Though any hints about properly dumping them would be appreciated (first time working with this kind of stuff).

Biktorgj, 2021-07-09:
You can do it from EDL mode. You will be missing some partition information you can later retrieve from any of the other firmware update packages, but at least you'll be able to dump all of the partitions. The stock kernel is hacked to prevent you from dumping some of the partitions.

Check out EDL tools from here: https://github.com/bkerler/edl
And run: `./edl.py rl dumps --skip=userdata --genxml` while in EDL mode in the modem

Zapeth, 2021-07-16:
> Check out EDL tools from here: https://github.com/bkerler/edl
> And run: `./edl.py rl dumps --skip=userdata --genxml` while in EDL mode **in** the modem

You mean this should be executed from the rootfs partition or from phone userspace (emmc/sd)?

In case of the former, whats the best/cleanest way to install the necessary requirements for `edl.py`? (python3, pip and dependencies).

In case of the latter, I'm guessing the [installation instructions](https://github.com/bkerler/edl#installation) are still applicable for the pinephone (distro differences notwithstanding)?

Biktorgj, 2021-07-17:
You need to run this from the Pinephone, it wouldn't fit in the modem :). Simply follow the installation instructions for linux and you're set
If running stock firmware, you can run `adb reboot edl` to force reboot into EDL mode, then run the command. It should take a while to download everything as EDL is relatively slow

Zapeth, 2021-07-17:
Thanks, I managed to dump the partitions now (which is basically all data that is listed [here](https://wiki.pine64.org/wiki/PineModems#NAND_Partition_table_layout).

It would be even better if the firmware files that were used for flashing the stock version were available, but I guess this should suffice (assuming its safe to use them for re-flashing?)

Also here's the sha-1 of my dumped `modem.bin`, maybe this can still be matched between different devices? -> 09353AF136AC401CABDFB572FDD3B0A4E4B320D7

Biktorgj, 2021-07-17:
If it has _never_ been mounted rw maybe, but don't think so since you're probably also dumping the empty space at the end of the partition. If you want to do that you're better just doing shasum's of the files themselves.

Since that firmware version is not available at my repo, can you share it with me so I can make a flashable package? I would need all files _except_ usr_data, misc and efs partitions

Biktorgj, 2021-08-04:
When you dump via EDL you bypass any limitation inserted by Quectel, so yes, you read everything, as you should being the owner of your device :)

Don't try (at least at this point with edl tools) to flash it back with EDL. I broke (without remedy) a modem this way when something went south and marked some sectors in the partition table as badblocks

Tomorrow will download your file and make a flashable package. Thanks!

Zapeth, 2021-08-05:
Thanks @AndreySV for providing your files, I haven't gotten around to find a suitable file host for uploading mine yet.

However comparing the dumped files, almost all partitions are identical anyway, so I guess the empty partition space didn't end up being an issue afterall (in regards to reproducability).

Biktorgj, 2021-08-05:
Okay, pushed!
I have only diverted from the default branch for the kernel, recovery, modem and system partition, as all the rest should be identical or almost identical, and I wouldn't want to mess around with bootloaders without reasons as this is first a recovery repo and second a firmware archive (don't want anyone bricking because they selected the wrong branch)

I also added some other firmware versions I had but hadn't published (in case you're curious)
https://github.com/Biktorgj/quectel_eg25_recovery/branches/all

closed 2021-09-25 by Biktorgj