Biktorgj/quectel_eg25_recovery #10: update EG25 via OpenWRT

dnwk, 2021-11-01:
I am wondering if I could run this update on an OpenWRT system that has EG25 attached via mini PCIe

ASoTNetworks, 2021-11-03:
This is what I want to figure out too

ASoTNetworks, 2021-11-04:
I figured it out you have to build https://github.com/Biktorgj/qfirehose in OpenWRT buildroot https://openwrt.org/docs/guide-developer/toolchain/crosscompile which will build qfirehose for your SoC. Then you can replace qfirehose with your own then run `./qfirehose -f ./` as in the readme file in this repo.

Biktorgj, 2021-11-04:
I left statically linked binaries for both arm32 and arm64, if your board is arm based you can probably use that directly, otherwise yes, you'd have to build qfirehose yourself (or manually flash them with fastboot if openWRT has it)

dnwk, 2021-11-04:
Thanks. Mine are MIPS. So do need to compile. I may try to connect my EG25 to a raspberry pi via USB and do it there.

Biktorgj, 2021-12-23:
Closing it due to inactivity, if there's something else feel free to reopen the issue

closed 2021-12-23 by Biktorgj