Biktorgj/quectel_eg25_recovery #11: Quirks Resolced

PsychoGame, 2021-11-22:
Hello @Biktorgj,

For those quirks I found the solution to this.
I was having a lot of issues as well on Manjaro ARM Phosh.
Stopping ModemManager through systemctl stops interference from ModemManager (Tries to re-initialize the modem again) and solves the Sahara flash failure:
`sudo systemctl stop ModemManager`
After successfully flashing the firmware you can restart ModemManager again.
If you could include this in the README, others might be helped out as well, as ModemManager in pmOS and Mobian will be updated eventually as well.

Greetings Jasper

nexero, 2021-12-22:
Works for me in pmOS :+1: 
```sh
sudo rc-service modemmanager stop
sudo ./qfirehose -f ./
sudo rc-service modemmanager start
```

thanks for the hint @PsychoGame

Biktorgj, 2021-12-23:
Thanks @PsychoGame!
Added the quirk to the readme: https://github.com/Biktorgj/quectel_eg25_recovery/commit/1351e026335d5b5fc395d232410388e082724d11

closed 2021-12-23 by Biktorgj