Biktorgj/quectel_eg25_recovery #12: Add new version `EG25GGBR07A08M2G_30.004.30.004`

Zapeth, 2022-01-18:
I was curious if any newer firmware versions were available and indeed stumbled on a new version on quectel's forum, `EG25GGBR07A08M2G_30.004.30.004`  is available from posted sharepoint link at: https://forums.quectel.com/t/eg25-firmware-and-lwm2m/12126

Has anybody tested this yet with modem on Pinephone (including custom firmware)?

Looking at the firmware release changes it seems version `EG25GGBR07A08M2G_30.003.30.003` released before that, maybe this can also be found/requested somewhere.

Biktorgj, 2022-01-19:
Hi and thanks for the link!
I haven't seen that firmware before, downloading it now...

Will give it a try and if there's nothing strange going on with it will upload it to the recovery repo (leaving this issue open until merged)

Biktorgj, 2022-01-21:
Tested the full image a bit and booted and reported just fine
The ADSP firmware seems stable too (that one I've been using it for the last two days and so far no issues).
New firmware is here: https://github.com/Biktorgj/quectel_eg25_recovery/tree/EG25GGBR07A08M2G_30.004.30.004

Thank you @Zapeth!

closed 2022-01-21 by Biktorgj