Biktorgj/quectel_eg25_recovery #15: volte (voice over lte) does not work with operator side data block

zylzyz, 2022-05-07:
volte (voice over lte) does not work with operator side data block.

in this issue i have used lte-only mode for devices. downgrading into 2g or 3g is not functional volte. operator data block is different than mobile data button in devices. mobile data button did not affect of volte to either direction.

i suspect issue being in the eg25-g chip but could be in the pinephone's operating system.

because i have tested another phone which is modified android 11, then i know volte works with or without operator data block.

in pinephone, if operator data block is activated, then volte does not work. i have also tried using AT commands without using UI and result is the same. i have tested 30.004.30.004 quectel's firmware and nearly opensource firmware. software mobile data button did not change to any direction.

how to i continue with this issue?

Biktorgj, 2022-05-13:
There could be a lot going on here.

VoLTE requires functional IMS profiles to work, and, depending on the carrier, the carrier needs to support the specific model to allow functional VoLTE (for example, Verizon). If the running IMS profile isn't working for VoLTE, the modem will automatically downgrade the call to standard Circuit Switch (normal GSM call) mode. 
First test, to see if it is actually related to available data service, would be to attempt a call with data enabled, and check if the icon in Phosh stays in 4G or changes to 3G/2G, then you can determine if the problem lies in the data session, or if the problem is a missing or mismatched IMS profile.

Quectel only has a limited set of IMS profiles in their modems (about 20), whereas a Oneplus 3T, for example, has about 60.

zylzyz, 2022-05-13:
maybe i was not clear but volte works with pinephone/eg25g, but only when operator blockage is not activated. profile is row_generic.

does volte work or not.

| | operator data block on | software data block off |
| -- | --- | --- |
| software mobile data on  | no | yes |
| software mobile data off | no | yes |

operator could be a problem but volte worked in custom android 11 in all cases, so at this time i don't think that operator is a problem. (for other operators, i have heard that operator block forbids volte in all cases).

i also tried at commands like AT+CREG AT+COPS AT+CLCC ATD; etc. in a way pinephone's u.i. option for mobile was disabled, same results.

question is more like, how i get more info like debug, logs and commands to know what's going on?

zylzyz, 2022-05-21:
it seems that something got messed up with my phone, another pinephone did not have this issue. i did some stuff around and manage to fix this. i still think something fishy is going on. essentially, volte works if operator's mobile data block is on. (operators sometimes implement mobile data block in a way that blocks volte). 

if you have issues with volte, either partial of full, maybe you should reinitialize volte settings, even if they are default or set.

```
AT+QMBNCFG="select","ROW_Generic_3GPP"
AT+QCFG="ims",1
AT+CFUN=1,1
```

factory reset may help some extreme situations.

```
!!warning!! only use these if modem chip is seriously messed up!
needs to be used as bundle
AT+QPRTPARA=3  < Factory reset NVM items (careful, you may lose modem control!)
AT+QPRTPARA=1  < Save NVM items
!!warning!!
```

you may close this issue now.