Biktorgj/quectel_eg25_recovery #16: PPP recovery qfirehose repeated failure

henrythemouse, 2022-05-08:
Hi,

I'm trying to use the instructions for installing the recovery firmware. It seems to go well enough until id doesn't. qfirehose repeatedly prints two lines that end in 'fail'. I am just typing commands, and mostly lost. Can anyone offer any help on this problem?

```
[002.276]: Totals checking 18 files md5 value, 0 file fail!
[152.999]: qfirehose.c main 366 fail

```
I've tried rebooting the phone, and stoping ModemManager, getting the  same output. The numbers at the start of the output lines do change, like [152.503] to [152.999].

```
lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 005: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[mobianModemFirmwareRecovery.log](https://github.com/Biktorgj/quectel_eg25_recovery/files/8647730/mobianModemFirmwareRecovery.log)

```

henrythemouse, 2022-05-09:
I finally gave up on this effort and installed 0.6.4, which took less than 30 seconds. So far this version seems to fix most issues.

For a while I didn't have any phone sound, after a couple reboots that came back. Now the phone is operational.

The modem is quite slow to come alive after suspend, actually the phone seems less responsive in general when returning from a suspend. It's just a matter of seconds, but .... push the power button and wait, tap on an app icon and wait ... I rebooted and although the modem came on I had no sound again.

Still, moving in the right direction I think. Looking forward to more improvements.

closed 2022-05-09 by henrythemouse

Biktorgj, 2022-05-09:
The numbers changing are the seconds it was running... But would need more detailed log to guess what went wrong :(
You also have a script called "flashstock" you can use to restore the modem to factory state.
If checking out the custom firmware, make sure you also change a couple of settings to make the modem perform as it should (for example, so it needs zero seconds to be functional after resume)
https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/SETTINGS.md#pinephone-pro

henrythemouse, 2022-05-09:
Thanks for the reminder, I had forgotten about these recommendations. Done now.