Biktorgj/quectel_eg25_recovery #20: interpretation questions

sbe-arg, 2022-09-22:
1- here: https://github.com/Biktorgj/quectel_eg25_recovery#flashing-with-fastboot the host is the pinephone?
2- if 1 is the pinephone and not a desktop distro how do you get fastboot working in postmarketos?
3- if 1 is a desktop distribution how does the AT command work at all? https://github.com/Biktorgj/quectel_eg25_recovery/blob/EG25GGBR07A08M2G_30.006.30.006/flashstock#L3
4- if 1 is a desktop and the AT command needs to be run on the pinephone we could update the documentation to reflect it in 2 steps.

I think my english as second language is not allowing me put the entire picture together.

mouffa, 2022-09-22:
@sbe-arg  the host is the Pinephone, that is the device where the modem is attached to

look how much easy it was for me

1) the modem had a SIM and was connected to the carrier in a fully working state
2) I downloaded the repository as a ZIP file
3) I unzipped it
4) I entered the folder
5) I ran the ./flashstock command
6)the modem disappeared for some seconds and came back again fully working with the new firmware

that's all

mouffa, 2022-09-22:
@sbe-arg I did it on Manjaro and I had installed a package called `android-tools` which contains fastboot, search for something similar in your repositories

sbe-arg, 2022-09-23:
I'm gonna try a different distro. Thanks for the comments @mouffa 
In postmarketos installed `android-tools` but I get fastboot command uknown, adb works wich is odd.

mouffa, 2022-09-23:
here it says it's in `/usr/bin`

https://pkgs.alpinelinux.org/contents?file=&path=&name=android-tools&branch=v3.16&repo=community&arch=aarch64

you can try reinstalling `android-tools`

sbe-arg, 2022-09-23:
Did some cleanup and reinstalled android-tools and worked.

Interesting. And thanks for the heads up explanation above!

sbe-arg, 2022-09-23:
Managed to revert the firmaware succesfully. Ill close the issue.

closed 2022-09-23 by sbe-arg