Biktorgj/quectel_eg25_recovery #22: Firmware changing?

adryb, 2023-01-24:
I have EG25-G module with firmware EG25GG**C**R07A02M1G, what mean **C** because your have B in name...

mouffa, 2023-01-25:
@adryb https://media.digikey.com/pdf/PCNs/Quectel/PCN_2022042905.pdf

adryb, 2023-01-25:
Ok, i found this document but i still dont know if i can use this firmware?

mouffa, 2023-01-25:
@adryb  on my modem's metal case is inscribed "EG25GGB-256-SGNS" and I have flashed the "EG25GGBR07A08M2G_30.006.30.006" firmware, it works properly

adryb, 2023-01-25:
i have: EG25GG**C** and wondering about this **C**

mouffa, 2023-01-25:
@adryb what is the model on the modem? there are many variants, B/C/D/L/GBTEA etc, and many different firmwares, all the modems seem to have the same datasheet and it may has to do with the country it can be sold because of the bands

adryb, 2023-01-25:
EG-25G means that is global, not country specific. I dont know what second letters means...

Look here:
https://cdn.shopify.com/s/files/1/2462/2517/products/4G-router_module_xe300_1000x1000.jpg?v=1672284642

mouffa, 2023-01-25:
The first “G” means “Global” and then it follows GB/GC/GD/GL/GBTEA etc, it may be something with the licensing or some technical detail that hasn’t to do with the functionality.

adryb, 2023-01-26:
Yes but i dont know if i can upload int GC firmware named GB...

mouffa, 2023-01-26:
@adryb ask them here https://forums.quectel.com/

adryb, 2023-01-26:
already did. :)

https://forums.quectel.com/t/difference-eg25ggc-vs-eg25ggb/19972

mouffa, 2023-01-26:
@adryb what kind of modem do you own? is it on a Pine64 device or is it a mini-pci express card?

adryb, 2023-01-26:
mini-pci