Biktorgj/quectel_eg25_recovery #25: Quectel EG25GGBR07A08M2G doesn't function after recovery

akamensh, 2024-02-17:
The Quectel modem EG25GGBR07A08M2G (bought in January 2024 with the PinePhone Pro Explorer Edition) stopped booting after an attempt to replace the stock firmware with the [community one](https://github.com/Biktorgj/quectel_eg25_recovery) via fwupd (v. 0.7.2) on Mobian 12. The recovery (QDL mode) returned it back to life, and it is possible to communicate to it from the system - but it can't be enabled. In particular the IMEI and the SN are not being retrieved. All the available firmware versions were tried with the same result. The log of the qfirehose: [qfirehose.log](https://github.com/Biktorgj/quectel_eg25_recovery/files/14319883/qfirehose.log) contains some "bad block" and "connection timeout" messages, while the final status is "Upgrade module successfully." Is there any way to deal with it?

Biktorgj, 2024-03-30:
The problem is the newer firmware from Quectel, it doesn't seem to allow downgrading and seems to have some incompatibility with the custom firmware that I haven't been able to find. You can downgrade though.
I assume you are still able to enter fastboot with `echo -ne "AT-QFASTBOOT\n\r" > /dev/ttyUSB2`

If that's the case, go here: https://themodemdistro.com/efs_recovery
And follow the instructions to recreate the EFS area, then it should start working again

Korv1n7, 2024-11-03:
> The problem is the newer firmware from Quectel, it doesn't seem to allow downgrading and seems to have some incompatibility with the custom firmware that I haven't been able to find. You can [...]

i upgrade 3 modem this instruction  ok

and one modem have trouble :(

root@alex-Aspire-3820:/home/alex/quectel_eg25_recovery-EG25GGBR07A08M2G_./flashstock
Sending AT+QFASTBOOT...
Sending OEM STAY in case the custom bootloader is installed
                                                   OKAY [  0.002s]
Finished. Total time: 0.002s
Waiting to settle...
Start flashing!
Sending 'aboot' (466 KB)                           OKAY [  0.025s]
Writing 'aboot'                                    OKAY [  0.176s]
Finished. Total time: 0.222s
Sending 'boot' (6066 KB)                           OKAY [  0.274s]
Writing 'boot'                                     OKAY [  2.070s]
Finished. Total time: 2.386s
Sending 'recovery' (6066 KB)                       OKAY [  0.237s]
Writing 'recovery'                                 OKAY [  2.071s]
Finished. Total time: 2.319s
Sending 'system' (56960 KB)                        OKAY [  2.215s]
Writing 'system'                                   OKAY [ 19.609s]
Finished. Total time: 22.056s
Sending 'recoveryfs' (9856 KB)                     OKAY [  0.512s]
Writing 'recoveryfs'                               OKAY [  3.446s]
Finished. Total time: 4.017s
Sending 'modem' (40064 KB)                         OKAY [  2.047s]
Writing 'modem'                                    OKAY [ 13.745s]
Finished. Total time: 15.975s
Rebooting                                          OKAY [  0.004s]
Finished. Total time: 0.055s
Done!

root@alex-Aspire-3820:/home/alex/EFS_Rec# fastboot flash aboot appsboot.mbn && fastboot reboot && fastboot oem stay
Sending 'aboot' (268 KB)                           OKAY [  0.015s]
Writing 'aboot'                                    OKAY [  0.114s]
Finished. Total time: 0.141s
Rebooting                                          OKAY [  0.004s]
Finished. Total time: 0.105s
< waiting for any device >
                                                   OKAY [  0.002s]
Finished. Total time: 0.002s
root@alex-Aspire-3820:/home/alex/EFS_Rec# fastboot flash sys_rev sys_rev.bin
Sending 'sys_rev' (5120 KB)                        OKAY [  0.280s]
Writing 'sys_rev'                                  FAILED (remote: 'flash write failure')
fastboot: error: Command failed

How fix this??

Thanks

baptx, 2026-04-03:
@Biktorgj Does it mean by following the instructions on https://themodemdistro.com/efs_recovery, it is possible to downgrade without issues below version EG25GGBR07A08M2G_30.200.30.200 even if the release notes says we cannot, "otherwise the module will not be able to work normally"?