Biktorgj/quectel_eg25_recovery #26: EG25GGBR07A08M2G_30.006.30.006: ADB and fastboot (still) not available

omasanori, 2024-07-22:
The situation is very similar to https://github.com/the-modem-distro/pinephone_modem_sdk/issues/99 but I decided to report anyways...

I have PinePhone 1.2a. The OS was updated to postmarketOS v24.06.

```shell
pine64-pinephone:~$ uname -a
Linux pine64-pinephone 6.9.0 #1-postmarketos-allwinner SMP PREEMPT Wed May 29 14:47:46 UTC 20 aarch64 Linux
pine64-pinephone:~$ cat /etc/os-release 
PRETTY_NAME="postmarketOS v24.06"
NAME="postmarketOS"
VERSION_ID="v24.06"
VERSION="v24.06"
ID="postmarketos"
ID_LIKE="alpine"
HOME_URL="https://www.postmarketos.org/"
SUPPORT_URL="https://gitlab.com/postmarketOS"
BUG_REPORT_URL="https://gitlab.com/postmarketOS/pmaports/issues"
LOGO="postmarketos-logo"
pine64-pinephone:~$
```

My modem had had firmware version `01.001.01.001` so I decided to update it to `EG25GGBR07A08M2G_30.006.30.006` by qfirehose in EDL mode first. I had tried the ADB method first but it didn't work. The test point method works then.

My expectation is that ADB and/or fastboot become working with that newer firmware, but it doesn't. By issuing `AT+QFASTBOOT`, the modem disappears and then comes back after few seconds, not as a fastboot device but a normal modem.

```shell
pine64-pinephone:~$ echo 'AT+QGMR' | sudo atinout - /dev/ttyUSB2 -

EG25GGBR07A08M2G_30.006.30.006

OK
pine64-pinephone:~$ echo 'AT+QFASTBOOT=?' | sudo atinout - /dev/ttyUSB2 -

OK
pine64-pinephone:~$ echo 'AT+QFASTBOOT' | sudo atinout - /dev/ttyUSB2 -
EOF from modem
pine64-pinephone:~$ # Few seconds later...
pine64-pinephone:~$ lsusb
Bus 003 Device 001: ID 1d6b:0001
Bus 002 Device 003: ID 2c7c:0125
Bus 001 Device 001: ID 1d6b:0002
Bus 004 Device 001: ID 1d6b:0001
Bus 002 Device 001: ID 1d6b:0002
pine64-pinephone:~$
```

Also, the `AT+QADBKEY` and `AT+QCFG` method with [qadbkey-unlock](https://xnux.eu/devices/feature/modem-pp.html#toc-unlock-adb-access) does not make ADB access available.

```shell
pine64-pinephone:~$ echo 'AT+QADBKEY?' | sudo atinout - /dev/ttyUSB2 -

+QADBKEY: ********

OK
pine64-pinephone:~$ ./qadbkey-unlock ********
AT+QADBKEY="****************"
AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,1,0

To disable ADB, run: (beware that modem will not be able to enter sleep with ADB enabled!!)
AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,0,0
pine64-pinephone:~$ echo 'AT+QADBKEY="****************"' | sudo atinout - /dev/ttyUSB2 -

OK
pine64-pinephone:~$ echo 'AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,1,0' | sudo atinout - /dev/ttyUSB2 -

OK
pine64-pinephone:~$ adb shell
* daemon not running; starting now at tcp:5037
* daemon started successfully
adb: no devices/emulators found
pine64-pinephone:~$
```

Any suggestions are welcome.

Zapeth, 2024-07-29:
I'm not sure if its the same issue, but ever since the pmOS update to v24.06 I also can't seem to get ADB working again.

It was definitely working on my modem with pmOS v23.xx (custom firmware installed), though its been a while since I last had to use it, so don't remember the exact steps that I did previously.

I tried sending `AT+QCFG="usbcfg",0x2C7C,0x125,1,1,1,1,1,1,0` followed by `AT+ADBON` and both return `OK`, but `adb devices` still prints `adb: no devices/emulators found`

Some noteworthy things:
- both `AT+QADBKEY?` and `AT+QCFG="usbcfg"` just return `OK`, but without any data (though I'm not sure if the latter should return anything)
- SMS commands to the custom modem FW also don't seem to work anymore since the upgrade
- the `AT+ADBON` command definitely does _something_ since the display brightness briefly flickers after sending it (modem restart?)
- with the pmOS upgrade the `android-tools` package was updated from v34.0.4 to v35.0.1, so maybe something major changed? (would be helpful to get feedback from someone using that version and having adb working)

I wonder if something might be interfering with the modem restart, like in https://github.com/Biktorgj/quectel_eg25_recovery/issues/18#issuecomment-1171892213? (though I already tried applying the same steps here, with no success)

Anyway now I'm also out of ideas, so I would also appreciate suggestions for further debugging.

omasanori, 2024-07-29:
Thanks @Zapeth, at least I realized that I am not alone...

I will try v23.12 to see if anything changes.

> both AT+QADBKEY? and AT+QCFG="usbcfg" just return OK, but without any data (though I'm not sure if the latter should return anything)

Mine returns something related (the serial number and the current configuration I guess). Interesting.

omasanori, 2024-07-30:
> It was definitely working with pmOS v23.xx (custom firmware installed),

Did you mean https://github.com/the-modem-distro/pinephone_modem_sdk was flashed to the modem, @Zapeth?

If so, I confirmed that `AT+QADBKEY?` and `AT+QCFG="usbcfg"` returns nothing but OK with that firmware just now. I think it is fine.

Arnavion, 2024-07-30:
(Reposting what I told omasanori on IRC...)

pmOS 24.06 onwards carries a udev rule added by [this change](https://gitlab.com/postmarketOS/pmaports/-/commit/d67aa6b44de452c1c5f14b8be07c1453a8e97ca1) to automatically restart the modem when it vanishes from USB. eg25-manager used to restart the modem itself when this happened, but that broke about a year ago, so pmOS does it itself by having udev restart eg25-manager and thus restart the modem.

So if you want to reboot the modem into fastboot, you don't want this udev rule to fire, so you have to mask it first.

1. Mask the udev rule.

    ```sh
    ln -s /dev/null /etc/udev/rules.d/90-modem-eg25-openrc.rules
    ```

2. Run `fastboot` or `flashall` or whatever you wanted to do.

3. Unmask the rule when you're done.

    ```sh
    rm /etc/udev/rules.d/90-modem-eg25-openrc.rules
    ```

Zapeth, 2024-07-30:
@Arnavion Thanks for the information, unfortunately it still doesn't work for me with the rule masked, the overall behavior seems unchanged (I also tried moving the file out of `/usr/lib/udev/rules.d` entirely).

I did notice that I had a `80-modem-eg25.rules` file beside the packaged `90-modem-eg25.rules`, maybe this wasn't correctly migrated during the pmOS upgrade? (I did modify the rules file at some point in v23.12, maybe thats why its still present).

To be safe I also moved this file out of the `rules.d` directory, but that didn't seem to change anything either.

I assume you're also on pmOS v24.06 and adb is working for you, can you share the steps you follow to get adb working on your device?

@omasanori Is adb working for you on v24.06 after masking `90-modem-eg25-openrc.rules`?

> If so, I confirmed that AT+QADBKEY? and AT+QCFG="usbcfg" returns nothing but OK with that firmware just now. I think it is fine.

Yes I have the custom firmware installed, thanks for checking the commands.

Biktorgj, 2024-07-30:
@Zapeth Two things: 
- QADBKEY only works on stock, if you're running the custom firmware you have to enable adb with AT+ADBON (No keys needed)
- You're trying to start adb as a normal user, please switch to root first, then, as root, kill adb if it was running (`adb kill-server`) and then try to run `adb devices`  again

Arnavion, 2024-07-30:
>I did notice that I had a `80-modem-eg25.rules` file beside the packaged `90-modem-eg25.rules`, maybe this wasn't correctly migrated during the pmOS upgrade?

There is nothing to migrate. Both of those files are expected to exist and do different things. The 80- file comes from aports `eg25-manager` package and the 90- from pmaports `device-pine64-pinephone` package.

>(I did modify the rules file at some point in v23.12, maybe thats why its still present).

Modifications to files in /usr do not matter to apk. Only modifications to files in /etc are preserved. In any case you should not modify files in /usr.

>I assume you're also on pmOS v24.06

I'm on Edge. (I'm the pmOS maintainer for the PinePhone.)

>and adb is working for you

I just checked and it isn't. Just like you saw, sending `AT+ADBON` reconnects the modem to USB but it doesn't appear reachable to `adb devices -l` (and yes I did `pkill adb` first to make sure the server gets restarted and redetects devices).

I haven't tried adb since I used it to switch from Quectel fw to OSS fw in 2021 or so, so I believe you that it broke between 23.12 and 24.06. The only thing I've done recently is use `fastboot` as part of upgrading the OSS firmware (`AT+QFASTBOOT` + `fastboot ...`) which still works fine, after the udev rule is disabled.

I posted what I posted because this part of the OP:

>By issuing `AT+QFASTBOOT`, the modem disappears and then comes back after few seconds, not as a fastboot device but a normal modem.

... is because of the udev rule I mentioned.

omasanori, 2024-07-30:
Thanks @Arnavion and @Biktorgj for insightful explanations. I learnt many.

> @omasanori Is adb working for you on v24.06 after masking 90-modem-eg25-openrc.rules?

No, on v24.06 with udev rules masking, adb does not work while fastboot does work.

Arnavion, 2024-07-30:
Looking at the changelog of android-tools between Alpine 3.19 and Alpine 3.20 versions (34.0.4 -> 35.0.1), I spotted there was a change for adb to use libusb by default in 35.0.1

https://developer.android.com/tools/releases/platform-tools

>35.0.0 (February 2024)
>
>- adb
>
>    - Switch to libusb as the default on Linux [issue #270205252](https://issuetracker.google.com/issues/270205252).

Reading through that issue, I spotted mention of an `ADB_LIBUSB` env var. So on a hunch I did `pkill adb; ADB_LIBUSB=0 adb devices -l` and now the modem shows up for me:

```sh
$ pkill adb; ADB_LIBUSB=0 adb devices -l
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
community_fw           device usb:2-1 transport_id:1
```

This is on Edge. Try it on 24.06

omasanori, 2024-07-30:
@Arnavion I confirmed that the ADB_LIBUSB workaround works on v24.06 too. (as I replied on chat)

OTOH I had built android-tools 23.0.4-r6 for v24.06 and confirmed that it works (without ADB_LIBUSB).

Arnavion, 2024-07-30:
Yes, 34.0.4 defaults to not using libusb, so `ADB_LIBUSB=0` is not needed for it. In fact, I would guess that `ADB_LIBUSB=1` will break it. You can try.

omasanori, 2024-07-30:
I agreed but actually adb 34.0.4 does work with ADB_LIBUSB=1, I just tried. Probably libusb on v24.06 handles its job as expected.

Arnavion, 2024-07-30:
Okay. In any case, something is wrong in android-tools or libusb or in the interaction between them, so this should be followed up in an aports issue.

omasanori, 2024-07-30:
Sure, would you open one on the aports issue tracker and put a link here? I will close this issue after that.

Arnavion, 2024-07-30:
https://gitlab.alpinelinux.org/alpine/aports/-/issues/16326

Zapeth, 2024-07-30:
I can also confirm that it works for me with again with `ADB_LIBUSB=0`, thanks for the quick help